wareya/DIY-Gaming-Mouse #2: Use optical switches [suggestion]

raszpl, 2024-04-21:
No debouncing required, infinite click lifetime, only limit is LED lifetime.
Buttons $2 [D2FP-FN2](https://www.digikey.com/en/products/detail/omron-electronics-inc-emc-div/D2FP-FN2/21410416)
Can also design your own PCB optical beam sensorts and have 3d printed button move slit into the path of light. Razer and TTC do a plastic shell with a lever going into PCB light sensors, here a 3d cutout of TTC http://en.ttc9.com/product/50.html fancy animation https://www.linkedin.com/posts/jeffrey-hu-8a5559101_ttc-optical-micro-switch-for-mouse-with-a-activity-6762816336910532608-CXmb

Wheel https://cdn4.explainthatstuff.com/how-ball-mouse-works.jpg Would need to hunt right size with good availability. Something like:
$3 https://www.ebay.com/itm/305117600440 [Photoelectric Speed Sensor Encoder Coded Disc Code Wheel For Freescale Smart gs] 
$4 https://www.ebay.com/itm/144130543656 [HC-020K Photoelectric Rotational Speed Sensor with Encoder Code Disc Code WheelH]

Mechanical encoders are just a cost optimization, part of Logitech plan to turn mouse business into Recurring Revenue stream with everyone replacing their mice once a year.

I put optical switches into my Steelseries to make it reliable. Im still looking for perfect scroll wheel mechanism. Especially click relying on rotating shaft directly pressing on the microswitch is a big fail, delivers uneven often squeaky rotation.
Personally I went reusing components route by buying defective, most often "double click", top shelf mice for peanuts (~$1 + shipping) and mixing/matching parts.

wareya, 2024-04-22:
Optical switches are something I'm totally down for looking into, but it might be quite a while before I research them. My backlog is stuffed.

For the 3d printed light blocking idea, I'm afraid filament 3d printers aren't quite precise enough for it. Mouse switches have to be precise to within something like 0.05mm, but the standard nozzle thickness on 3d printers is 0.4mm.

Optical encoders actually have some challenges of their own. Because they're exposed to open air, it's easy for them to gunk up and start skipping or glitching on specific sections of the rotation. But at the same time, they're easier to clean non-destructively and keep working for a very long time. IMO, they're a tradeoff.

wareya, 2024-04-22:
PS: My personal solution to M3/middle click button problems is to make the dpi button a second middle mouse button.