wareya/DIY-Gaming-Mouse #3: Question: Height of PCB?

C4illin, 2024-05-31:
What height should the PCB be? 1.6mm?

wareya, 2024-05-31:
Yep, 1.6mm standard!

C4illin, 2024-05-31:
Thank you!

Do you have any JLCPCB part list? The project.db seems to be unpopulated

wareya, 2024-05-31:
It seems to contain the right data for me:

[image]

The plugin that uses project.db is https://github.com/Bouni/kicad-jlcpcb-tools

C4illin, 2024-06-01:
Ohh didn't know about that plugin. It was only missing R40 but that one was easy to find :) Thanks for the help!

closed 2024-06-01 by C4illin

wareya, 2024-06-01:
Make sure you don't do what I did and accidentally get a 3.9k ohm resistor instead of a 39 ohm one! The 39 ohm ones don't show up in the 'Basic' catalog.

C4illin, 2024-06-01:
Yeah if it was something I learned from the video it was that 😄
I actually used that frame in the video to try matching so I got all component right, before asking here :)

I was thinking of taking the C137630 from extended catalog.

wareya, 2024-06-01:
Yep, the C137630 should work!

C4illin, 2024-06-01:
Thanks for all the help!

C4illin, 2024-06-02:
Hello, in the DB it says to use C22775 for R3, R4 but C22775 is 100 ohm and not 27 ohm as specified, what is correct?

wareya, 2024-06-02:
Those would be the USB data line impedance control resistors. I think putting in 100 ohm resistors is probably a mistake on my part.

Their exact resistance probably doesn't matter too much. 27.4 ohms is the value suggested/specified by the rp2040 hardware design docs. I would expect anything between 10 and 50 to work in practice. The target resistance for the entire data line according to the usb spec is 90 ohms, and real usb parent devices are more generous than that in practice.

I'll amend the part selection and double check everything again when I have a few spare hours.

C4illin, 2024-06-02:
100 ohm is much cheaper on jlcpcb ;)

wareya, 2024-06-02:
Looks like I did indeed end up using 100 ohm resistors (01A):

[image]

So yeah, should still work! Unless you have a particularly picky machine.