wareya/DIY-Gaming-Mouse #4: Recreation

SpillAway, 2024-06-18:
Hey Wareya,

So basically I'm working with a team of friends trying to recreate your mouse. We have the PCB which we got from PCB Way. I was wondering if you could walk us through the process. You can contact me by email at: pn52tfzvy@mozmail.com.

Thank You
Kind Regards 
Spill_Away

misel228, 2024-06-18:
There's a very detailed list of things you need to know be capable of before this endevour, right on the start page: https://github.com/wareya/DIY-Gaming-Mouse

Read this and start working on it. When you get stuck, try to explain what exactly is happening *and* what you expect instead.

When asking questions, try to follow this guide: https://stackoverflow.com/help/how-to-ask

Good luck

wareya, 2024-06-18:
The above post is right. The best place to start with this kind of activity is by trying and getting it wrong, and then figuring out how to fix what went wrong. If you run into any specific problems, feel free to ask! I'll be happy to help. The more specific you are and the fewer back-and-forths it takes, the better my help will be.

SpillAway, 2024-06-26:
We have a **$300** budget and we need to make 3. So we can't fail and then expect to do it again.

wareya, 2024-06-26:
The shell parts weren't designed in traditional CAD software, they were designed in blender. The .blend file is the source file. There are no STEP source files.

A 300 dollar budget sounds like plenty to make a few mistakes, as long as that budget doesn't include getting the tools needed to build and assemble everything (3d printer, etc).