wareya/DIY-Gaming-Mouse #5: Tuning mouse report rate for optimal VSync frame pacing

fuweichin, 2024-08-21:
To avoid pointer movement stuttering when coalescing HID input messages as VSync-aligned pointer events, it is necessary to design a mouse report rate (or say sampling rate, sometimes) which is integral multiple of display refresh rate.

So, for the common refresh rates 60Hz/120Hz, it's neccesary to use a report rate like 120Hz to keep VSync frames even in terms of pointer movement.

Yet I cannot find a mouse which is optimized for VSync frame pacing, that's why I'm interested in DIY/open-source mouse.

Why integral multiple? Here are some testing results with [Pointer Movement Timing Analyzer](https://beta.tourofdreamland.net/device-tests/hid/pointer-movement-timing-analyzer/):

Figure1: using 119Hz report rate with 60Hz refresh rate MAY cause 1 jumpy frame per second
[image]

Figure2: using 125Hz report rate with 60Hz refresh rate MAY cause 5 jumpy frames per second
[image]

Apple continuously optimizes report rates for VSync frame pacing, and now only leaves its Magic Mouse and Magic Trackpad behind.

Table1: Apple Products HID Sampling Rate

| Device                 | Pointer Type   | Sampling Rate | Refresh Rate |
| ---------------------- | -------------- | ------------- | ------------ |
| iPhone 13 Pro          | touchscreen    | 240Hz         | 120Hz        |
| iPhone 12              | touchscreen    | 120Hz         | 60Hz         |
| iPad Pro 12.9" 2nd gen | touchscreen    | 120Hz         | 120Hz        |
| Apple Pencil           | pen            | 240Hz         |              |
| MacBook Pro M1         | mouse/touchpad | 120Hz         | 120Hz        |
| Magic Mouse            | mouse          | 90Hz          |              |
| Magic Trackpad         | mouse/touchpad | 90Hz          |              |

wareya, 2024-08-21:
USB 2.x HID report rates unfortunately need to be millisecond-aligned, which means that the only options between 125hz and 1000hz are as follows:

```
125hz (8ms)
~142.8hz (7ms)
~166.6hz (6ms)
200hz (5ms)
250hz (4ms)
~333.3hz (3ms)
500hz (2ms)
1000hz (1ms)
```

Any report rates for a USB HID device between these is, in reality, switching back and forth between similar rates to give an average of the measured rate.

The easiest way to avoid input device movement microstutter is to use as high of a report rate as possible. This is one of the reasons that gaming mouse firmware (including this repo) has decided on 1000hz (sometimes 500hz) as the standard. Doing this also minimizes input latency. Modern office mice still use 125hz, but they're just being weird/bad.

Some modern mice are capable of using even higher sampling rates, like 8000hz, over higher-speed versions of USB. For these mice, more sample rates are possible, but for input latency reasons, it's still best to use the highest sample rate that can be stably maintained. The rp2040 isn't capable of these even higher-speed versions of USB.

Also, the 3060 sensor in this mouse has a sampling rate of its own (something in the dozens of khz), so at some point down the signal chain, there's always going to be a small amount of microstutter.

If you're on linux, you can sometimes override USB device polling rate: https://wiki.archlinux.org/title/Mouse_polling_rate#Set_polling_interval

fuweichin, 2024-08-21:
Mouse DIY players can take the lead in improving mouse VSync frame pacing even via some just-works solution, as a proof of concept to deliver unique value to the industry, e.g. use 8000Hz (0.125ms interval) sampling rate sensor then downsample to a of 120Hz (8.333...ms interval) report rate, e.g. ~119.4Hz (8.375ms interval), which is better than 125Hz in terms of VSync frame pacing, better than 1000Hz in terms of energy efficiency.

That shouldn’t be something only-Apple-can-do.

Indeed, 1000Hz report rate can reduce VSync pointer movement microstutter,
[image]

but is at the cost of higher CPU usage, e.g. moving 1000Hz mouse over empty area causes 10% CPU usage(i7-7920HQ, Windows 10), on Windows 10 desktop that causes extra 10% CPU usage if “Show desktop icons” is checked, which means more heat and fan noise.
[image]