wareya/DIY-Gaming-Mouse #6: More details needed about ripping setup

Deozaan, 2024-09-10:
Thanks for the video and the open repository. I'm trying to follow along but a simple photograph of the ripping setup isn't adequate for me. The angle is not great and it is difficult for me to understand what needs to be connected to what.

Would it be possible to get a written description of the setup? Which colored wires in the photo are connecting to which pins on the Pico? What type of resistor is on the breadboard, and which pins on the Pico is it connected to? And are there any labels on the PWM sensor/chip to make it explicit which ones are being connected to?

I'd really prefer not to fry any of my components so I'd like to be able to triple check that everything is as it should be before attempting this. Thank you!

wareya, 2024-09-10:
Following this image as reference: https://i.imgur.com/EspAlvz.jpeg

The sensor has the text "A23" on the "bottom" edge, i.e. towards where the wrist would be when holding the mouse, the top of the photo. However, the sensor is usually mounted upside-down, so this is actually the top.

[image]
(https://i.imgur.com/zneMFco.png)

This illustration shows the pin numbers for the sensor and which pin numbers need to be tapped to record the SPI message.

- GPIO16 need to be connected to pin 10
- GPIO17 need to be connected to pin 11
- GPIO18 need to be connected to pin 12
- GPIO19 need to be connected to pin 13
- GPIO19 needs to be pulled high to a 3.3 volt source with a 1 ***mega***-ohm resistor. The exact value is not important but it should be very large. 10kohms for example would be too low. 100kohm to 10mega-ohms is probably a safe range.

Deozaan, 2024-09-10:
Thank you very much for the clear and detailed explanation.