wareya/DIY-Gaming-Mouse #8: Wireless plans?

bam80, 2024-12-08:


wareya, 2024-12-08:
currently doing the planning 👍 
[image]

bam80, 2024-12-08:
Hi Alexander,
that's great to hear!

I'm interested with the firmware, did you considered one?
I found another project (no wireless so far) based on QMK:
https://github.com/qmk/qmk_firmware/tree/master/keyboards/ploopyco

But maybe ZMK is a better suit for wireless?
What do you think?

wareya, 2024-12-08:
I'm going to run another version of my own custom firmware so that I can micromanage power delivery and sleep states.

bam80, 2024-12-08:
Good luck!

willwaush, 2025-01-19:
Are you planning on using the PicoW for wireless connectivity?

Would be glad to help in terms of electronics if needed. Can't wait for this project to get the wireless functionality!