wareya/DIY-Gaming-Mouse #10: BOM missing part R39

BrendanSiche, 2025-02-04:
Hello,

I would like to order the pcb, I generated the files using kicad and the jlcpcb plugin
I'm not sure about the last row tho.
The value of R39 is R40 in the comments, is that just a cosmetic thing or have the row been mismatched somehow ?
I tried looking for the part itself(R_0603_1608Metric) but everything seems to be out of stock

Does anyone know what that last row should be before I order the pcb ?

additional question, but I know the video build log had an issue with a wrong value resistor beeing ordered (3.9k ohm instead of 39) is it fixed in the repo's file ?

Thanks !

[image]

wareya, 2025-02-04:
40 ohms is acceptable and close enough, I think I had problems finding an R39 at the time and tried replacing it with an R40, but somehow confused myself and ended up with a 3.9k. I don't remember whether I fixed it in the repo yet or not.

See also: https://github.com/wareya/DIY-Gaming-Mouse/issues/3#issuecomment-2143805471

BrendanSiche, 2025-02-04:
Thanks !
I chose C137630 for the missing part, it seems to be good at 39 ohm, ordering the pcb now, I hope it'll work out okay !