wareya/DIY-Gaming-Mouse #12: Issue with SROM ripping code.

josht000, 2025-09-14:
I have the RP2040 all connected as you and others shared. However, I had to modify the code just a little to get to compile and such. I'm not getting any output in the Serial Monitor and want to veryify a few things in the code. The [gist](https://gist.github.com/wareya/0dc3498bd80be93c47b6f79aa9164321) you point us to says, "long SPI message recorder - rasberry pi pico (rp2040), arduino IDE .ino file (C++)" at the top of the page. Ok Great, so it's an Arduino IDE project. Well then why does it include  #include <pico/stdlib.h>??? Everyone says you shouldn't do this and indeed I get file not found errors. I had to use #include <Arduino.h> instead and replace Serial.printf with printf(). Could that be causing me issues by not seeing output on the Serial Monitor?? What did you do? 

Thanks for any help and my PCB's are already in the states on their way to me :)

josht000, 2025-09-14:
I know those solder globs are **sus,** but there's no connection there - _probably_. 
[image]

wareya, 2025-09-14:
It's strange that you needed to change the includes, and it's probably the reason that it's not printing anything, yeah. Which board target configuration were you using?

josht000, 2025-09-14:
RP2040 Pico 1 using the Arduino IDE and the MBed OS RP2040 Boards, by Arduino, version 4.4.1 installed.

What version did you use to program? 

[image]

josht000, 2025-09-14:
@wareya v4.4.1?

josht000, 2025-09-16:
Ok, my options look the same as yours and the code is essentially running fine, but still not able to get the full 4096 bytes out. Would you mind sharing where you got your donor mouse?

wareya, 2025-09-16:
It was "Plugable"'s pmw3360 mouse and I bought it from amazon, but it's no longer for sale unfortunately.

If you don't specifically need to rip the SROM code yourself, you can dummy out the SROM uploading parts of the firmware and the mouse should still work; it'll just be more liklely to glitch out on incompatible tracking surfaces.

It might somehow be a problem with wire length or solder connection quality, but it would be really weird if this was the problem. Maybe I forgot about a buffer to ground or live voltage or something...

Deozaan, 2025-09-16:
I also had trouble dumping/ripping the SROM in full when I attempted this about a year ago. I never got it to work in full, all at once. I kept getting fragments of varying lengths, always different, but with sections of sequences of numbers that matched other fragments from other attempts.

After many attempts I was eventually able to piece the fragments together into a full 4096 byte file. I was able to verify that I had a match to a known-good SROM file by finding other DIY mouse repositories which included 4096 byte SROM files for this sensor and comparing them to what I had pieced together.